mcqn/my-babys-got-led #7: Build a mounting board

amcewen, 2020-11-08:
Given ATX power supplies have a standard set of mounting points, we could laser-cut a nice board that mounts to the power supply and then provides suitable mounting points for the PCB, to keep everything together more neatly.

arthurrowland, 2021-05-10:
Given we've seen that there is quite a lot of variation in ATX mounting, shall we adapt this issue to be just for mounting the board itself, and the user can adapt to their own supply and situation?

arthurrowland, 2021-06-14:
I prepared this one-side board with two layers of 4mm MDF and M3x12 bolts. Would need etching to be useful. It's quite satisfying to hold and for casual applications the molex cable works fairly was as a stand to hold it at a useful angle. Don't love it, but is quite inexpensive to produce.

[image]

arthurrowland, 2021-06-21:
Following an in-person discussion with @amcewen, I have been looking at the many PSU mounting patterns to scope out how many we might want to draw up, if any, beyond the 'standard' ATX mounting pattern.

Here's a really good summary of the different types of ATX power supplies, with the mounting patterns and external dimensions clearly visible:
http://www.evercase.co.uk/index.php?route=information/information&information_id=7

Here are the most common power supplies stacked, with an adaptor plate on the top SFU-type PSU.

[image]

Here's a drawing for the cutout for SFX type PSUs found [here](https://www.overclock.net/threads/official-usff-sff-club.1300645/page-76):

[image]

and here's the ATX one while we're at it, since it's not already linked in this issue:

[image]

- Option 1: No mcqn support for SFX type
- Option 2: Design mounting plate with both sets of holes
- Option 3: Supply separate mounting plate for SFX
- Option 4: Supply adaptor plate for SFX to ATX - for example [£4 on ebay](https://www.ebay.co.uk/itm/154474553329) and [£2 on alibaba](https://www.aliexpress.com/item/1005002011841397.html)
- Option 5: Flag mounting issues and link people to adaptor plates <- most pragmatic option in my view

Next steps are to update the small board protector to become a sandwich-style case and add some simple outline labels, and update the PSU panel mount version with the latest, larger, revision of My Baby's Got LED.

arthurrowland, 2021-07-13:
I have [renamed](https://github.com/mcqn/my-babys-got-led/commit/b23c36c1570613539b88990fc99c7ff27c7890fe) the ATX PS2 mounting board files ([twice](https://github.com/mcqn/my-babys-got-led/commit/a32d96c2b49a5abdc0e383e5d23d10f2a5657efc)) and included the design files for the following enclosure (rough prototype from scavenged acrylic, but dimensions are correct).

[image]

It uses 20mm standoffs on the molex side and 10mm standoffs on the LED/status light side.

The new file names distinguish between rev1 and rev2 my baby's got LED boards and the PSU mounting patterns (where needed).

closed 2021-07-27 by arthurrowland