mcqn/my-babys-got-led #9: Apply for OSHW certification

amcewen, 2020-11-08:
Given it's open hardware, we should get the OSHWA certification for it.  https://application.oshwa.org/apply

amcewen, 2021-02-10:
What license is the documentation under?  Having run through the CC licence chooser, it will be [CC BY-SA 4.0](http://creativecommons.org/licenses/by-sa/4.0/?ref=chooser-v1).  We should add the documentation to the Github repo.

Do we need to change licence for the board itself (i.e. is AGPL not the right choice - CERN, Solderpad or TAPR are the named choices on the OSHWA application)? https://certification.oshwa.org/process/hardware.html gives a brief overview of those hardware licences.

From that guide, the *rough* mapping from software -> hardware is GPL -> CERN; Apache -> Solderpad; GPL -> TAPR (which can also cover documentation).

Reading through it, the CERN-OHL-S v2.0 is the most applicable, so I've updated the licence for that.

arthurrowland, 2021-05-10:
May as well open source the whole application process, reproducing the application form currently hosted at this URL: https://application.oshwa.org/apply

- [ ] @arthurrowland drafted
- [ ] @amcewen updated/approved
- [ ] @arthurrowland applied

**BASIC INFORMATION
SECTION 1 OF 4**
This part of the form asks for basic information about the responsible party for the project to be certified, such as the name of the individual, company, or organization certifying the project and contact information for any future correspondence from OSHWA regarding your certification.

This certification is on behalf of a: Company
Name of Individual, Company, or Organization Responsible for the Certified Item: MCQN Ltd
If not an Individual, name of Individual with Authority to Bind the Company or Organization: Adrian McEwen

**Contact Information**
Country: United Kingdom
Street Address Line 1: DoES Liverpool, The Tapestry
Street Address Line 2: 68-76 Kempston Street 
City/town/village: Liverpool
State/Province/Region: Merseyside
Zip code/postal code: L3 8HL
OSHWA Contact Email Address: redacted
Public contact email address: info@mcqn.com

**PROJECT INFORMATION
SECTION 2 OF 4**
This part of the form asks for information about the project you wish to certify. Information submitted here will appear on your project’s profile listing.

Project name: My Baby's Got LED
Project version: 1.0
Have you previously registered a version(s) of your project with OSHWA? No
Project description: My Baby's Got LED is an LED driver board using WLED to drive individually addressable LEDs at 5V, using an ATX power supply. It uses an ESP8266 chip and can be controlled using its own WiFi network or by connecting to an existing network. The development and documentation for the board is available on github, and the board is manufactured by mcqn and sold on Tindie.
Project website: mcqn.com/ibal223
Primary Project Type: IOT
Additional Project Types: Arts, Electronics, Home Connection
Project Keywords: WLED, MQTT

Where can the documentation be found for your project?: https://github.com/mcqn/my-babys-got-led

Does your project incorporate or build upon other open projects that are not currently certified by OSHWA? If so, use this space to cite those projects

| Citation Title | Citation URL |
| --------------- | --------------- |
| WLED | https://github.com/Aircoookie/WLED |

**LICENSING INFORMATION
SECTION 3 OF 4**
This part of the form asks for information about your project’s licensing. In order to qualify for OSHWA certification, you must have chosen an open source license for your hardware, your software (if any), and your documentation. Your licenses for each of these elements will appear on your project’s profile listing.

What license is your project's hardware licensed under? CERN
What license is your project's software licensed under? @amcewen is this GPL?
What license is your project's documentation licensed under? CC BY-SA

| Statement | Response | Explanation |
| --------------- | --------------- | --------------- |
| The project is licensed in a way to allow for modifications and derivative works without commercial restriction. | Yes | |
| There is no restriction within my control to selling or giving away the project documentation. | Yes | |
| Where possible, I have chosen to use components in my hardware that are openly licensed. | Yes | |
| I understand and comply with the "Creator Contribution requirement," explained in the [Requirements for Certification](https://certification.oshwa.org/requirements.html). | Yes | |
| There is no restriction on the use by persons or groups, or by the field of endeavor. | Yes | |
| The rights granted by any license on the project applies to all whom the work is redistributed to. | Yes | |
| The rights granted under any license on the project do not depend on the licensed work being part of a specific product. | Yes | |
| The rights granted under any license on the project do not restrict other hardware or software, for example by requiring that all other hardware or software sold with the item be open source. | No | CC-BY-SA requires adaptions of the documentation use the same license. @amcewen what about the other licenses? |
| The rights granted under any license on the project are technology neutral. | Yes | |

CERTIFICATION
SECTION 4 OF 4
This part of the form asks you to certify that you agree to specific terms of the [OSHWA Certification Mark License Agreement](https://certification.oshwa.org/license-agreement).

You agree to the following terms:
- [x] I have provided OSHWA with accurate contact information, recognize that all official communications from OSHWA will be directed to that contact information, and will update that contact information as necessary.
- [x] I will only use the certification mark in compliance with official certification guidelines.
- [x] I acknowledge that all right, title, and interest in the certification mark remains with OSHWA.
- [x] I acknowledge that OSHWA has the right to enforce violations of the use of the mark. This enforcement may involve financial penalties for misuse in bad faith.
- [x] I have the ability to bind those responsible for the certified item to this agreement.

If you do not agree with any of the above terms, please explain.
Provide a brief explanation (500 characters)
Relationship to certified item: I am the primary developer of the certified item, as well as a manufacturer and retailer of the item as a commercial product.

[... cut ...] ntity listed to the License Agreement. I recognize that I will receive my unique identification number that allows me to promote my project as OSHWA Open Source Hardware Certified in compliance with the user guidelines via the email provided to OSHWA after submitting this form.

If you are the parent or legal guardian entering into this agreement on behalf of an individual under the age of 18, please provide your name to certify that you also agree to be bound by this agreement.: Not applicable.

amcewen, 2021-05-10:
According to [the LICENCE file in WLED](https://github.com/mcqn/WLED/blob/master/LICENSE) it's MIT.

So the software can be used in other non-open projects, but the hardware is restricted under the CERN Open Hardware Licence Version 2 - Strongly Reciprocal.

It's not v1.0.  Checking the board, we've only got a revision number.  I should maybe have included a version number, but at present I think `r20210224` is the best version indicator we've got.

We probably need to document the documentation licence somewhere...

amcewen, 2021-05-10:
LGTM :+1:

arthurrowland, 2021-05-10:
[image]

arthurrowland, 2021-05-24:
We did it!

https://certification.oshwa.org/uk000030.html

[image]

closed 2021-05-24 by arthurrowland