mcqn/my-babys-got-led #12: Rework the Electric Liverbird in DoES Liverpool to use a My Baby's Got LED board

amcewen, 2021-02-01:
The [Electric Liverbird](https://mcqn.com/ibal145/) in DoES Liverpool currently has a Raspberry Pi and Arduino and custom power supply to drive its Neopixels.

It'd be a good candidate to convert to a My Baby's Got LED board to show what's possible, and use the segment support for different parts of the bird: beak; newspaper; legs; head; wing...

Then we could get the MQTT integration working to animate it more in DoES Liverpool.

amcewen, 2021-02-12:
The Electric Liverbird has WS2801 LEDs in it, so requires a special build of `WLED` with `USE_WS2801` defined.  We got it upgraded today and it's up and running.  There's a bit of tidying up still to do:
 * [ ] Replace the fuse with a 10A, because there are only 150 LEDs in it, so it won't need any more
 * [ ] Crimp bootlace ferrules onto the wires in the screw terminals
 * [ ] Box up the board and power supply
 * [ ] [BONUS] Provide a 5V supply for the LED matrix display, given Julian had been powering it from a spare USB port on the old setup

amcewen, 2021-02-26:
Looks like there's a new feature in WLED (I'm pulling the latest changes in at the moment to update our fork) which [might make it easier to map the Liverbird LEDs to useful segments](https://github.com/Aircoookie/WLED/commit/2544d2e0680d4fd27ae3741bb773d9e513e42b0d)

arthurrowland, 2021-05-10:
We still achieved a decent outcome even with the board floating loose: https://mcqn.com/posts/doestheliverbirdplay-connecting-twitter-to-wled-to-find-out/

Shall I box it up tomorrow in a simple birch ply thing? Any unusual specifications?

amcewen, 2021-05-10:
That sounds like a plan.  It'd be good to etch the photo of the Post and Echo building (featuring the Liverbird in its original home) onto the front of the case again, with the "The Electric Liverbird" and credits text.  The original electronics box design is in the LiveRGBird repo on the MCQN server.

I've ordered [one of these USB A sockets](https://uk.farnell.com/tuk/kua2whpm/usb-conn-2-0-type-a-rcpt-wht/dp/3367486) too, so we can provide a 5V power supply for the coffee age display.

amcewen, 2021-05-10:
IKR.  I would've bought one with a better datasheet, but didn't find any at a reasonable price.  It should turn up tomorrow anyway, so shouldn't slow things down too much.

I don't have any strong opinions on box generation.  Whichever you prefer.

arthurrowland, 2021-06-29:
I used the [universal box generator](https://festi.info/boxes.py/UniversalBox) from boxes.py with the following settings, for nominal 4mm MDF and an ambitious burn setting.

[image]

It was very tight going together, probably because the material was actually 4.22mm, though that would have already been factored into the [burn test](https://festi.info/boxes.py/BurnTest) I ran before cutting the main box.

I seem to have lost my touch when it comes to etching - poor results on both MDF and Plywood (not pictured - had hoped to put a nice cut-out panel of the main image on top of the MDF).

Anyway it's a workable enclosure, but I couldn't get the Liver bird to power on using the terminals. Figured it would be quicker overall to wait for @amcewen input rather than poke about with it.

[image]
[image]
[image]

arthurrowland, 2021-06-30:
We're almost there!

I replaced the side panel and fitted the same rev1 board back into position.

[image]

There's a video of part of the build here: https://twitter.com/MCQN_Ltd/status/1410218247572668419

Do we want to publicly document how to play with it using node-RED, here or over on the DoES github? Both? Neither?

@amcewen where is the matrix display mentioned early in this issue? Strictly this issue is just to provide the port, but it feels like we need to plug it in to close it off properly :)

closed 2021-07-05 by arthurrowland