mcqn/my-babys-got-led #18: Create some tutorials on how to use MBGL/WLED to do interesting things

amcewen, 2021-05-09:
In the same way that we [recorded a LiveCoding session for connecting the Electric Liverbird to Twitter](https://www.youtube.com/watch?v=6vbz_RKSmfA, we should make some more guides to how to use the MBGL boards to do cool stuff.

We can note possible ideas here...

* [ ] Using https://github.com/scottlawsonbc/audio-reactive-led-strip to build a WLED VU-meter
* [ ] Building on that last one, setting up a Pi as a dedicated VU-monitor (would be handy in the [DoES Liverpool](https://doesliverpool.com) event room, hooked up to the PA...)
* [ ] A tide clock, so you can pick the right time to go walk the dog
* [ ] Ditto but so you can catch the sunrise/sunset ([this seems a decent JSON API for the Liverpool times, for example](https://api.sunrise-sunset.org/json?lat=53.4143&lng=-2.9807)

amcewen, 2021-05-10:
Maybe try out https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options as an alternative option for sound-reactive that doesn't require an additional computer/Pi (but does need a microphone/line-in circuit to be wired into the board)

arthurrowland, 2021-05-10:
Uh oh, I've had an idea... A vision of a warm, dusky summer evening gig in the park by a friendly band, with some IP rated LED strips surrounding the band reacting to the music. Whether we want to resource that vision is another thing! If the gig was 3 hours long and the board was drawing 24A (restricted by the fuses), we get a requiredment of 24,000mAh which is within reach of something [like this 3-pin power bank](https://www.ebay.co.uk/itm/184460937085) if we can borrow one. Day trip for the geometric lights? Or best kick this into the someday/maybe pile?

amcewen, 2021-05-10:
Hehe, well... on Friday's ride Danny was pitching his plans for some short gigs in them sometimes, which did lead me into wondering about some lights for it.  I don't think it's a priority, but might be worth a bit more back-and-forth over the coming months.

arthurrowland, 2021-05-10:
Even better! Will let it simmer.

amcewen, 2021-08-23:
Create a few #Cheerlights-based MQTT channels, so people can easily set their board up to do cheerlights at Christmas