mcqn/my-babys-got-led #20: Write up how to update the firmware on your board

amcewen, 2021-06-04:
Some of our users will want to update the software, so we should write a short guide (or point them at the relevant bit of the WLED wiki).

There'll be two options:
 * Using the OTA update to do it over WiFi (we should test that it works too - I've not tried it yet)
 * Using an FTDI (or similar) cable

We'll need to publish releases of our version of WLED (the only difference being the couple of usermods for the status LEDs and factory reset option) for people to be able to easily get a new version, rather than having to build their own (but I guess we should add a "build from source" section to the reflashing guide too)

arthurrowland, 2021-08-18:
Using [this .bin file (zipped)](https://github.com/mcqn/my-babys-got-led/files/7008084/my-babys-got-led-WLED-20210818.zip), here are the steps I followed to successfully update the board I'm using for the Totaliser. Unfortunately it failed while I was writing this documenation... so the steps only go so far (though that is most of the way).

Download the `.bin` file ready on the a device with access to WLED - so on the same WiFi network, whether that is the WLED-AP or another network.

Open WLED controls page

[image]

Open WLED's `Config` menu

[image]

Open `Security and Updates`

[image]

Scroll down to the `Software Update` section and click `Manual OTA Update`

[image]

From this screen, click choose file and select the `.bin` file you saved in the first step

[image]

When your file is shown, click `Update` and stay on the webpage

[image]

Wait for it to go through these screens while updating:

[image]

If it fails, you will get this message - please contact us on info@mcqn.com to let us know it's gone wrong :/

[image]

huffeec, 2022-05-23:
I have managed to update to the local build of v0.13.1 using FTDI but not using OTA.
(But v0.13.1 doesn't seem to drive the lights?)

Once installed it's possible to use the OTA tool to write the local v0.11.1 or v0.13.1 bins, but not the versions from the main branch of WLED.

But if the v0.11.1 version is then flashed OTA it's not possible to use OTA to update to v0.13.1

I'm not sure why OTA isn't working on v0.11.1 but this suggest to me that FTDI is the only route to updating existing boards atm?

Is there value in updating the documentation to use FTDI, will users have access to USB to serial devices?

huffeec, 2022-05-23:
Eg
1.  Install esptool.py as described here https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html
2.  Download the latest build of the My Baby's got LED firmware
3.  Connect your board to USB using a USB to serial connector and the pins marked 'Serial' on your board.
4.  Set the board to programming mode by holding down reset, using a jumper wire to connect P0 to ground and releasing the reset button
5. From the command line run 
`esptool.py --before default_reset --after hard_reset --chip esp8266 --port "/dev/ttyACM0" --baud 115200 write_flash 0x0 path/to/my-babys-got-led.bin`
NB Depending on your setup and OS the port description may be different

huffeec, 2022-05-25:
I've added a draft updatingFirmware.odt document but not referenced it anywhere yet. I will make these changes when the v0.13.1 bin is complete.

amcewen, 2022-06-15:
Digging into @huffeec's comment above:
>  (But v0.13.1 doesn't seem to drive the lights?)

This is because v0.13 removed the default relay pin, so the software doesn't turn on the power to the lights.  We should define it (to the old default setting of `12`) in our builds, and it looks like it might get configured as default when upgrading from earlier versions with the official WLED build(? Not 100% sure of this, but that seemed to be the case when upgrading one of our old boards over USB with the official v0.13.1)

If you end up with the wrong config, it can be fixed by...
1. Go into `LED Preferences` in config
2. Set `Relay GPIO` to 12, and make sure the `Invert` checkbox is ticked
3. `Save` those settings
4. Go back to the main screen
5. Turn the `Power` off and then back on (this seems to be needed to pick up the setting)

It all seems to be working now (I had a problem with the config getting corrupted, which I traced as far as the LittleFS `open` call failing, but an erase and reflash fixed things).

To use the stock WLED you need to choose one of the 2MB ESP8266 versions, so  `WLED_x_ESP02.bin`.  Wonder if that's why it fails on earlier builds of ours?

amcewen, 2023-11-15:
The OTA upload is working okay with the `ESP02.bin` versions until you get *beyond* `0.14.0-b1`.  At present any newer ones fail to upload - I'm not sure why at the moment.