mcqn/my-babys-got-led #21: Fabricate programming/test rig

amcewen, 2021-06-28:
Spinning this out of #15 so we can knock off more self--contained issues...

Now that we've got the basics of the programming rig set up, we should laser-cut an https://tinylabs.io/openfixture/ jig to hold the board.

I'll need pogo pin points for the following connections:
 * T1
 * T2
 * 5V SB
 * TX
 * RX
 * GND
 * 0
 * PWR_ON_T1
 * 5V_T1
 * Data_T1
 * Clk_T1
 * 12
 * 13
 * 14

We might not use all of them right from the start, but if they're included then all the testing options will be open.

amcewen, 2021-07-26:
We should re-cut the top of the base, to sort out the minor (~1mm) alignment issue, then glue in the pogo pins.

If that fixes the flakiness of the programming (after I tested it initially last week) then this can be closed.

arthurrowland, 2021-07-28:
After some wild goose chase cutting new inserts, I noticed there was some slack in the interlocking construction of the rig, and pushing it to the bottom edge got things to line up pretty well!

The only pogo pin still looking a little off is the PWR_ON_T1 hole, so I'll drill the top panel out a little so there's enough play to hit the target when glued into place.

After that I'll bounce it back to @amcewen for testing with the hope we can close this issue.

arthurrowland, 2021-07-28:
Bouncing!

amcewen, 2021-08-11:
We've got all the pins we need for now working.  We can open a new issue if need be (or fold the fixing into whatever issue of testing that it relates to).

closed 2021-08-11 by amcewen