mcqn/my-babys-got-led #23: Check why the mDNS name doesn't seem to get set up properly

amcewen, 2021-08-06:
This might be a bug in WLED, but raising it here for some initial investigation.  Setting the board's name in the settings doesn't always result in `name-you-chose.local` working to connect to the board.

arthurrowland, 2021-08-06:
Does [this WLED issue](https://github.com/Aircoookie/WLED/issues/762) or [this comment by aircoookie](https://github.com/Aircoookie/WLED/issues/1227#issuecomment-705108528) provide any clues?

Is the issue zeroconf support?

amcewen, 2021-08-13:
Have been digging into this off and on all week.  It *seems* that it's only a problem with programming it with Arduino rather than Platform IO.  An easy way to check if that's the problem would be if it also forgets any WiFi details if the board gets reset or power-cycled.

Older versions will be okay, but ones programmed with the latest versions of WLED (since it switched away from storing settings in EEPROM).

Programming it with the new rig is fine.  And it should be possible to upgrade the older ones over WiFi - I've not done that yet, but it would be useful to test and document.  We can get an updated `.bin` file from the programming rig, and then try the [instructions on the WLED wiki](https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary)

Can I throw that over to you to try, @arthurrowland

arthurrowland, 2021-08-18:
I've been trying to get mDNS to work on windows with no luck, so far! Working assumption is that it is all ok on the board but Windows and Android don't support it out-of-the-box. Would be good to figure it out and let people know how to get it working.

arthurrowland, 2021-08-20:
I will uncritically install obscure software to try and solve a problem, and just tried [this software](https://hobbyistsoftware.com/bonjourbrowser), which initially found the IP address of the board I was using. But it didn't on the second attempt, and also didn't pick up the other boards on the network (such as the one connected to the electric liver bird). So not recommended. Could be operator error but it seems you're just supposed to open it up and refresh.

On a sort-of-related-but-probably-another-issue note, it would be great if [this node red module, wled2](https://flows.nodered.org/node/node-red-contrib-wled2), [github here](https://github.com/danecreekphotography/node-red-contrib-wled2), supported mDNS too - bit out of scope for me but would make guides easier to document since we wouldn't have to do a 'find out your board's IP address' step.

amcewen, 2021-08-20:
Yeah, that's the same Windows app I'd been using to look for instances - it seemed to work reasonably well, but I've also got Apple's Bonjour service installed on that Windows machine.

huffeec, 2022-02-16:
How often is "doesn't always"?
I've tried renaming mDNS through settings and not managed to recreate it exactly (I don't think)

Changing the name in settings alone doesn't take effect - old_name.local works but new_name.local doesn't
After a software power cycle the same - old_name.local works but new_name.local doesn't
Resetting the board via the push button does seem to work consistently** -new_name.local works but old_name.local doesn't

**I did have some confusion when after a power cycle old_name.local and new_name.local both seemed to work. Tested across different browsers and I think this is caching .

amcewen, 2022-02-16:
Maybe it's more of a behaviour-isn't-what-you'd-expect issue then?  In that it only changes the name after a power-cycle.

How did you do the "software power cycle"?  Was that with the power button on the UI?  I *think* that's just to turn the LEDs off and on, rather than reset the whole board... (again more of a UI issue but more of a bug in the upstream WLED stuff)

huffeec, 2022-02-16:
Yes to ui "software power cycle" which explains why it didn't have the same effect. Should I add this to the user instructions and then close this for now?

amcewen, 2022-02-21:
That sounds like a good plan.

closed 2022-04-11 by huffeec